plapointe6/EspMQTTClient #102: Support WIFI_AP_STA

kintel, 2022-02-01:
I want to use this library together with ESP-NOW (I'm building an ESP-NOW to MQTT gateway). This works fine, except the WiFi mode needs to be `WiFi.mode(WIFI_AP_STA);` for this to work.

I do appreciate the low-config approach of this library, but can you think of an elegant way of making such advanced settings configurable?

EdJoPaTo, 2022-02-01:
I assume this should be possible with the changes of fffcd55faf284f1fe4295576d5bb346746ff8f7c (which are currently in the dev branch and not yet released)?

kintel, 2022-02-02:
Thanks,  I took the dev branch for a spin and it actually works out of the box!
The key to success is that `EspSimpleWifiHandler` doesn't touch the WiFi mode as long as the STA flag is already set.

closed 2022-02-02 by kintel