plapointe6/EspMQTTClient #111: Always connecting to same wifi AP with poor signal instead of one in the same room

aptalca, 2022-07-21:
## Description of the problem
I'm encountering this issue with the following project that relies on this library: https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32

I have 3 APs running on wifi channels 1, 6, and 11 with the same ssid, but the board always connects to the AP running on channel 1, even though the one running on channel 11 is in the same room as the board. It results in having really poor signal (-80 dBm). I tried kicking it off the network to force a reconnect, as well as rebooting multiple times. It insists on connecting to the one on channel 1.

## Versions
plapointe6/EspMQTTClient@^1.13.3

## Hardware
- Board type : ESP32
- Board model : HiLetgo ESP32S

## C++ code
https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/blob/main/PlatformIO%20Files/SwitchBot-BLE2MQTT-ESP32/src/SwitchBot-BLE2MQTT-ESP32.cpp#L2122

Is there anyway to define a mac address or even a channel for the wifi connection?

Ideally, the client would check the different APs on the same ssid and attempt to connect to the one with the strongest signal. Not sure if that's implemented here and if so whether there is a bug or not.

Thanks

PS. Cross-posting from https://github.com/devWaves/SwitchBot-MQTT-BLE-ESP32/issues/87

EdJoPaTo, 2022-07-21:
This library is a convenience wrapper around the official ESP8266/ESP32 WiFi methods so it only does what it does.
https://github.com/plapointe6/EspMQTTClient/blob/bd829e1352bc970b6fa6a99c7243d9205e2ef700/src/EspMQTTClient.cpp#L559-L571

Maybe the ESP32 WiFi supports what you are looking for? If that's the case this library could also support that. Otherwise its probably a feature request for the ESP32 WiFi library. I am relatively certain that there are already similar requests but I haven't checked anything there.

But I am curious what you might find out there.

aptalca, 2023-03-01:
Just to circle back on this, I went down the rabbit hole and ended up at a repo with no issue reporting (iirc it was some sort of a firmware repo by the manufacturer) and kind of hit a dead end.

As a workaround, I realized (or rather was informed by a colleague) that unifi controller has the option to lock a client to a specific AP, which is what I did.

I'll go ahead and close this issue.

Thanks

closed 2023-03-01 by aptalca