plapointe6/EspMQTTClient #112: How to configure for anonymous broker connection?

Jibun-no-Kage, 2022-08-04:
How to configure for anonymous broker connection?  I set user and password in configuration_t, but getting errors?

EdJoPaTo, 2022-08-04:
providing the error might help finding the cause. omitting user and password is possible for sure. You can also set them to NULL.

This is basically my minimal setup:

```c++
EspMQTTClient client(
  "WifiSSID",
  "WifiPassword",
  "hostname",       // MQTT Broker server ip
  "TestClient"      // Client name that uniquely identify your device
);
```

Jibun-no-Kage, 2022-08-05:
When you say in the documentation can be omitted, that means NOT PASSED IN, not set to NULL.  The function explicitly requires the user and password be defined, if NULL is required.  Suggest someone change the documentation.

Jibun-no-Kage, 2022-08-05:
Odd, that I did should have worked...

EspMQTTClient::EspMQTTClient(
  const char* wifiSsid,
  const char* wifiPassword,
  const char* mqttServerIp,
  const char* mqttClientName,
  const short mqttServerPort) :
  EspMQTTClient(wifiSsid, wifiPassword, mqttServerIp, NULL, NULL, mqttClientName, mqttServerPort)

I qualified ssid, passphrase, broker ip, client id, then port, but it generates and error.

no matching function for call to 'EspMQTTClient::EspMQTTClient(const String&, const String&, const String&, const String&, const int&)'

EspMQTTClient theClient(
  NETWORK,
  PASSWORD,
  BROKER,
  ID,
  PORT
);

First 4 are string constants, and port is a in constant.

EdJoPaTo, 2022-08-05:
The method is expecting a `const char*` and not a `String`. (Basically `C` vs `C++`)

You can convert the `String` to a c style `const char*` with `c_str()` like this: `something.c_str()`.

EdJoPaTo, 2022-08-05:
Personally I use `#define BROKER "hostname"` instead of String constants. This has the benefit of concating multiple ones at compile time like this:

```c
#define DEVICE_NAME something
#define BASE_TOPIC DEVICE_NAME "/status/"
```
BASE_TOPIC will be `something/status/` and thats done on compile time so no runtime String logic to be done.

for all the rest I prefer `const` Variables over defines (like the `PORT`)