plapointe6/EspMQTTClient #118: Can't access RabittMQ broker

sroumieux, 2022-11-05:
## Description of the problem
When I try to publish to my RabittMQ broker, I get :
_unable to connect (18.511000s), reason: MQTT_DISCONNECTED_

## Versions

- ESPMQTTClient lib: 1.13.3
- PubSubClient lib: 2.8.0
- Arduino ESP32 / ESP8622 core: 1.8.19 (To update it go to the arduino board manager and update ESP32 / ESP8266)

## Hardware
- Board type : ESP32
- Board model : ESP-WROOM-32D

## Logs
RabittMQ port is 5672

## C++ code

EdJoPaTo, 2022-11-06:
RabbitMQ is not MQTT. But it has an MQTT plugin: https://www.rabbitmq.com/mqtt.html

It’s default Port is 1883 (which is the common MQTT TCP port). Unless you configured 5672 in the MQTT plugin, you are probably connecting to something which is not an MQTT broker?

sroumieux, 2022-11-06:
It works now !  Thanks for your help !!
Stéphane

closed 2022-11-06 by sroumieux