plapointe6/EspMQTTClient #120: Configuration parameters for MQTT get lost over time (overwritten?)

mwildbolz, 2023-01-05:
## Description of the problem
I just saw, that the set MQTT parameters ( `client.setMqttServer(xxx)` ) during setup() Routine get lost/modified during subsequent calls of loop().
Maybe this is something in combination with the WifiManager library?
I stripped out parts of the code which are not relevant from my point of view. If everything is ok, I can share the complete code...

For example: The configured server IP (192.168.1.209) is first reported from the client correctly after configuration:
15:55:00.257 -> MQTT client initialized!
15:55:00.257 -> esp8266-PowerBar1
15:55:00.257 -> **192.168.1.209**

also my String variables are correctly filled:
15:55:00.257 -> MQTT serverIP: '192.168.1.209'
15:55:00.257 -> MQTT username: 'powerBar'
15:55:00.257 -> MQTT password: 'esp8266'

Later on it is changed to another value (in this case a debugging string, but I saw quite different characters here already):
15:55:00.779 -> 1/0/0 --> **1/**

Also the username is different!

What I'm doing wrong? Do you have any thoughts??

Thanks,

MW

## Versions

- ESPMQTTClient lib: 1.13.2
- PubSubClient lib: 2.8.0
- Arduino ESP32 / ESP8622 core: 3.0.2

## Hardware
- Board type : ESP8266
- Board model : AZ-Delivery D1 mini 12-F (ESP8266MOD)

## Logs

```
15:54:55.792 -> mounted file system
15:54:55.792 -> reading config file
15:54:55.792 -> opened config file
15:54:55.792 -> {"mqttServer":"192.168.1.209","mqttUser":"powerBar","mqttPwd":"esp8266"}
15:54:55.792 -> parsed json
15:54:55.792 -> *wm:AutoConnect 
15:54:55.792 -> *wm:Connecting to SAVED AP: WildWolf
15:54:56.305 -> *wm:connectTimeout not set, ESP waitForConnectResult... 
15:55:00.257 -> *wm:AutoConnect: SUCCESS 
15:55:00.257 -> *wm:STA IP Address: 192.168.1.179
15:55:00.257 -> Connected.
15:55:00.257 -> Starting MQTT config!
15:55:00.257 -> MQTT serverIP: '192.168.1.209'
15:55:00.257 -> MQTT username: 'powerBar'
15:55:00.257 -> MQTT password: 'esp8266'
15:55:00.257 -> MQTT client initialized!
15:55:00.257 -> esp8266-PowerBar1
15:55:00.257 -> 192.168.1.209
15:55:00.257 -> 0/0/0 --> 0/0/0 --> 
15:55:00.257 -> WiFi: Connected (4.596000s), ip : 192.168.1.179 
15:55:00.779 -> 1/0/0 --> 1/
15:55:00.779 -> MQTT: Connecting to broker "1/0/0 --> 1/" with client name "esp8266-PowerBar1" and username "���?" ... (5.102000s)unable to connect (5.127000s), reason: MQTT_CONNECT_FAILED
15:55:00.779 -> MQTT: Retrying to connect in 15 seconds.
15:55:00.826 -> MQTT!: Failed MQTT connection count: 1 
15:55:01.294 -> 1/0/0 --> 1/
15:55:01.808 -> 1/0/0 --> 1/
```

## C++ code

```c++

#include <FS.h> //this needs to be first, or it all crashes and burns...
#include <WiFiManager.h>          // https://github.com/tzapu/WiFiManager
#include <ArduinoJson.h>          // https://github.com/bblanchon/ArduinoJson
#include <Adafruit_NeoPixel.h>
#include <EspMQTTClient.h>
...
EspMQTTClient client;
...
void setup() {
...
  String smqttServer(mqttServer);
  String smqttUser(mqttUser);
  String smqttPwd(mqttPwd);
  const char* cpMqttServer = smqttServer.c_str();
  const char* cpMqttUser = smqttUser.c_str();
  const char* cpMqttPwd = smqttPwd.c_str();

  client.setMqttClientName(deviceName.c_str());

  if (smqttUser.length() > 0) {
    #ifdef DEBUG
      Serial.print("MQTT serverIP: '"); Serial.print(cpMqttServer); Serial.println("'");
      Serial.print("MQTT username: '"); Serial.print(cpMqttUser); Serial.println("'");
      Serial.print("MQTT password: '"); Serial.print(cpMqttPwd); Serial.println("'");
    #endif 
    client.setMqttServer(cpMqttServer, cpMqttUser, cpMqttPwd);
  } else {
    #ifdef DEBUG
      Serial.println("MQTT config without username.");
    #endif 
    client.setMqttServer(cpMqttServer);
  }

	client.enableOTA("ota"); // Enable OTA (Over The Air) updates. Password defaults to MQTTPassword. Port is the default OTA port. Can be overridden with enableOTA("password", port).
	client.enableLastWillMessage(statusTopic.c_str(), "offline", true);  // You can activate the retain flag by setting the third parameter to true
  #ifdef DEBUG
    client.enableDebuggingMessages();
  #endif
  
/* If I put this lines in, I get a connection to MQTT, but the data is also overwritten (so, later re-connection is not possible)
  delay(500);
  client.loop();
  delay(500);
  client.loop();
  */

  #ifdef DEBUG
    Serial.println("MQTT client initialized!");
    Serial.println(client.getMqttClientName());
    Serial.println(client.getMqttServerIp());
  #endif
}

void loop() {
  client.loop();
  delay(500);
}

```

mwildbolz, 2023-01-05:
Another log example (in this case, with the initial delayed calls of client.loop() within the setup routine). Here one can see, that the values are changing:

```
16:11:04.362 -> Connected.
16:11:04.362 -> Starting MQTT config!
16:11:04.362 -> MQTT serverIP: '192.168.1.209'
16:11:04.362 -> MQTT username: 'powerBar'
16:11:04.362 -> MQTT password: 'esp8266'
16:11:04.861 -> WiFi: Connected (5.083000s), ip : 192.168.1.179 
16:11:05.354 -> MQTT: Connecting to broker "192.168.1.209" with client name "esp8266-PowerBar1" and username "powerBar" ... (5.587000s) - ok. (5.600000s) 
16:11:05.354 -> MQTT client initialized!
16:11:05.354 -> esp8266-PowerBar1
16:11:05.354 -> 192.168.1.209
16:11:05.354 -> 1/0/0 --> b168.1.209
16:11:05.354 -> Connection established...
16:11:05.354 -> StatusTopic: esp8266-PowerBar1/status
16:11:05.354 -> MQTT << [esp8266-PowerBar1/status] online
16:11:05.354 -> MQTT: Subscribed to [openhab/ExcessPower]
16:11:05.400 -> MQTT: Subscribed to [openhab/PVPower]
16:11:05.400 -> MQTT: Subscribed to [openhab/BattSOC]
16:11:05.884 -> 1/1/1 --> l���?@
16:11:06.397 -> 1/1/1 --> -b�?
16:11:06.898 -> 1/1/1 --> -l�?
16:11:07.411 -> 1/1/1 --> #~���?��
```

EdJoPaTo, 2023-01-05:
Something I could think of (but dont know for sure) is changing content of the string types. Technically Strings and these char* thingies are memory areas where stuff can be written to.
If I understand your code right the String thingy is defined within the setup method. Then a pointer onto the String memory is created and passed to the library. Then the setup function is left, the variables can be removed and freed. The String is gone then but the pointer to that still exists. Other stuff might grab memory and just use that. When the still existing pointer is used it points into memory used by something but it isnt the non existing String anymore. Which fails.

I have not tested anything. This is just a theory I came up with when reading your issue briefly.

Personally I try to avoid String and char arrays and use defines for string type content which is known on compile time. That way I know its baked into the program memory and is fix and I can concat multiple strings at compile time without runtime logic being executed.

So something I use looks like this:

```cpp
#define WIFI_SSID "whatever"
// more defines
#define CLIENT_NAME "devicethingy"

EspMQTTClient mqttClient(
    WIFI_SSID,
    WIFI_PASSWORD,
    MQTT_SERVER,
    MQTT_USERNAME,
    MQTT_PASSWORD,
    CLIENT_NAME,
    1883);
    
#define BASE_TOPIC CLIENT_NAME "/"

void setup() {
  // setup other stuff too

  mqttClient.enableDebuggingMessages();
}

void onConnectionEstablished() {
  // will publish on the topic "devicethingy/git-version"
  mqttClient.publish(BASE_TOPIC "git-version", GIT_VERSION, MQTT_RETAINED);
}
```

Quickly got that example from my [esp-mqtt-dht](https://github.com/EdJoPaTo/esp-mqtt-dht).

Another way might be to create the Strings globally or static so they wont be removed when the setup method is finished.

Maybe that helps?

mwildbolz, 2023-01-06:
Thanks for the hint. The problem was in conjunction with the string parameter handling - more or less a copy and paste problem from the part with the configuration management.

I have rewritten this parts and now it is working, also when I have not hard coded the MQTT parameters..

closed 2023-01-06 by mwildbolz

Sperryfreak01, 2023-04-02:
@mwildbolz any chance you could publish your updated code? I am struggling with the same issue.

mwildbolz, 2023-04-02:
> @mwildbolz any chance you could publish your updated code? I am struggling with the same issue.

No problem. At the moment I use the following code, working well:
```
String deviceName = "esp8266";
String statusTopic = deviceName + "/status";
String IpTopic = deviceName + "/IP";

...

// MQTT parameters
String smqttServer;
String smqttUser;
String smqttPwd;

// Set web server port number to 80
WiFiServer server(80);

//flag for saving data
bool shouldSaveConfig = false;

EspMQTTClient client;

//callback notifying us of the need to save config
void saveConfigCallback () {
  Serial.println("Should save config");
  shouldSaveConfig = true;
}

void setup() {  
  Serial.begin(115200);

  //clean FS, for testing
  //SPIFFS.format();

  //read configuration from FS json
  #ifdef DEBUG 
    Serial.println("mounting FS...");
  #endif
  
  if (SPIFFS.begin()) {
    #ifdef DEBUG
      Serial.println("mounted file system");
    #endif
    
    if (SPIFFS.exists("/config.json")) {
      //file exists, reading and loading
      #ifdef DEBUG 
        Serial.println("reading config file");
      #endif
      
      File configFile = SPIFFS.open("/config.json", "r");
      if (configFile) {
        #ifdef DEBUG 
          Serial.println("opened config file");
        #endif
        
        size_t size = configFile.size();
        // Allocate a buffer to store contents of the file.
        std::unique_ptr<char[]> buf(new char[size]);

        configFile.readBytes(buf.get(), size);
        DynamicJsonBuffer jsonBuffer;
        JsonObject& json = jsonBuffer.parseObject(buf.get());
        
        #ifdef DEBUG 
          json.printTo(Serial);
        #endif

        if (json.success()) {
          #ifdef DEBUG 
            Serial.println("\nparsed json");
          #endif
          smqttServer = json.get<String>("mqttServer");
          smqttUser = json.get<String>("mqttUser");
          smqttPwd = json.get<String>("mqttPwd");
        } else {
          #ifdef DEBUG 
            Serial.println("failed to load json config");
          #endif
        }
      }
    }	
  } else {
    #ifdef DEBUG
      Serial.println("failed to mount FS");
    #endif
  }
  //end read
  
  // WiFiManager
  // Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //set config save notify callback
  wifiManager.setSaveConfigCallback(saveConfigCallback);
  
  // set custom ip for portal
  //wifiManager.setAPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

  //add all your parameters here
  WiFiManagerParameter custom_mqttServer("server", "MQTT Server", smqttServer.c_str(), 20);
  WiFiManagerParameter custom_mqttUser("user", "MQTT Username", smqttUser.c_str(), 10);
  WiFiManagerParameter custom_mqttPwd("password", "MQTT Password", smqttPwd.c_str(), 10);
  wifiManager.addParameter(&custom_mqttServer);
  wifiManager.addParameter(&custom_mqttUser);
  wifiManager.addParameter(&custom_mqttPwd);
  
  // fetches ssid and pass from eeprom and tries to connect
  // if it does not connect it starts an access point with the specified name
  // here  "AutoConnectAP"
  // and goes into a blocking loop awaiting configuration
  wifiManager.autoConnect("AutoConnectAP");
  
  // if you get here you have connected to the WiFi
  #ifdef DEBUG
    Serial.println("Connected.");
  #endif
  
  // Copy parameters into variables
  smqttServer = custom_mqttServer.getValue();
  smqttUser = custom_mqttUser.getValue();
  smqttPwd = custom_mqttPwd.getValue();
  
  //save the custom parameters to FS
  if (shouldSaveConfig) {
    #ifdef DEBUG 
      Serial.println("saving config");
    #endif

    DynamicJsonBuffer jsonBuffer;
    JsonObject& json = jsonBuffer.createObject();
    json["mqttServer"] = smqttServer;
    json["mqttUser"] = smqttUser;
    json["mqttPwd"] = smqttPwd;
    File configFile = SPIFFS.open("/config.json", "w");
    #ifdef DEBUG
      if (!configFile) {
        Serial.println("failed to open config file for writing");
      }
    json.printTo(Serial);
    #endif
    
    json.printTo(configFile);
    configFile.close();
    //end save
  }
  

	// MQTT
  #ifdef DEBUG
    Serial.println("Starting MQTT config!");
  #endif
  
  client.setMqttClientName(deviceName.c_str());

  if (smqttUser.length() > 0) {
    #ifdef DEBUG
      Serial.println("MQTT serverIP: '" + smqttServer + "'");
      Serial.println("MQTT username: '" + smqttUser + "'");
      Serial.println("MQTT password: '" + smqttPwd + "'");
    #endif 
    client.setMqttServer(smqttServer.c_str(), smqttUser.c_str(), smqttPwd.c_str());
  } else {
    #ifdef DEBUG
      Serial.println("MQTT config without username.");
    #endif 
    client.setMqttServer(smqttServer.c_str());
  }

	client.enableOTA("ota"); // Enable OTA (Over The Air) updates. Password defaults to MQTTPassword. Port is the default OTA port. Can be overridden with enableOTA("password", port).
	client.enableLastWillMessage(statusTopic.c_str(), "offline", true);  // You can activate the retain flag by setting the third parameter to true
  #ifdef DEBUG
    client.enableDebuggingMessages();
  #endif

  #ifdef DEBUG
    Serial.println("MQTT client initialized!");
    Serial.println(client.getMqttClientName());
    Serial.println(client.getMqttServerIp());
  #endif

	...
}
```