plapointe6/EspMQTTClient #123: ESP is Crashing/rebooting when using Webserver.h library

intoku, 2023-01-30:
## Description of the problem

For my application i need to run Webserver.h on port 80 on the ESP32. The program is compiling properly - until "server.begin();" is called - afterwards the program compiles - but crashes when uploaded to ESP.
Have tested the identical code with the standard library WiFi.h, where it works properly.
I have tried with including and not including the Webserver.h - mostly same issue. Furthermore, disabling or fully removing the client.enableHTTPWebUpdater(); did not help,

Is there a way to use the Webserver.h from my script without this conflicts/errors with the EspMQTTClient library?

Other tries:
- When using this script only with WifiClient.h or WiFi.h, and  - it works good 
- When the EspMQTTClient is included, but i use WiFi.begin(LOCAL_SSID, LOCAL_PASS); inside the setup() - it also works good
- When the EspMQTTClient is included, but i use EspMQTTClient client() - compiles but does not work
- when server.begin() is enabled - the whole application crashes and reboots

## Versions

- ESPMQTTClient lib: -1.13.3
- PubSubClient lib: -put version number here-
- Arduino ESP32 / ESP8622 core: ESP32-s2-rc4

## Hardware
- Board type : ESP32 or ESP8266 ?
- Board model :ESP32-s2-rc4

## Logs

```
Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40026d81
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x524
load:0x4004c000,len:0xa70
load:0x40050000,len:0x292c
entry 0x4004c18c

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace:0x4002735a:0x3ffd31000x4002c881:0x3ffd3120 0x40031c75:0x3ffd3140 0x400a9ea9:0x3ffd3270 0x400b8da5:0x3ffd32a0 0x400b8df6:0x3ffd32c0 0x400a9721:0x3ffd3310 0x4008bd12:0x3ffd3330 0x4008bd8c:0x3ffd3380 0x4008f469:0x3ffd33a0 0x40083593:0x3ffd33c0 0x40093332:0x3ffd3420 

ELF file SHA256: 0000000000000000

```

## C++ code

```c++
#include "EspMQTTClient.h"
//#include <WiFiClient.h>
#include <Wire.h>
#include <WebServer.h>
#define LOCAL_SSID "************"
#define LOCAL_PASS "************"

WebServer server(80);

  EspMQTTClient client(
    LOCAL_SSID, LOCAL_PASS, 
    "***********",  // MQTT Broker server ip
    "**********",   // Can be omitted if not needed
    "**********",   // Can be omitted if not needed
    "random_name",     // Client name that uniquely identify your device
    1883              // The MQTT port, default to 1883. this line can be omitted
  );

void setup(){
  //WiFi.begin(LOCAL_SSID, LOCAL_PASS);
  //while (WiFi.status() != WL_CONNECTED) { delay(500); }
  Serial.begin(115200);
  client.enableDebuggingMessages();
  
  server.on("/get", ShowGET);
  // server.begin();

  Serial.print("IP address: "); Serial.println(WiFi.localIP());
}

void onConnectionEstablished() {
    client.subscribe("WAVE/500/status/#", [](const String & topic, const String & payload) {
      String new_payload=payload;
    });
}

void loop()
{
  client.loop();
  server.handleClient();
}

void ShowGET() {
  Serial.println("works");
  server.send(200, "text/plain", "");
}

```

intoku, 2023-01-30:
Accidentally have found that if "server.begin();" is placed inside the "onConnectionEstablished()" function - than the server works. Not sure if this is the correct way to do it - can be helpful to anyone that needs the webserver active

github-ccnt-nib, 2023-04-25:
Hey folks. Ran into the same issue. Wanted to run my own webserver alongside the built-in http-update server without success, getting the boot-loop as shown above.
I solved the issue by exposing the webserver which is already built into the esp32mqttclient class. 
Inside EspMQTTClient.h add:
`WebServer* webserver();`
under the public: section (e.g. under enableHTTPWebUpdater())
Inside EspMQTTClient.cpp add:
`WebServer* EspMQTTClient::webserver() {
  return _httpServer;
}`

Then you can move the http-update url to some path, eg. 
`client.enableHTTPWebUpdater("/update"); `
and add all hooks, you want for your own webserver like so:
`client.webserver()->on("/", handle_root);`
the handle_root function needs to look like: 
`void handle_root() { 
  client.webserver()->send(200, "text/plain", "your answer");
}`