plapointe6/EspMQTTClient #128: Malformed UTF-8

benbecke, 2023-02-24:
## Description of the problem
From time to time when the esp restarts or reconnects to wifi i get the following message on mqtt broker when the client wants to connect. Seems that the client id is somehow malformed !?

1676553646: New connection from 192.168.195.60:56476 on port 1883.
1676553646: Bad socket read/write on client <unknown>: Malformed UTF-8

This is taking some time with this error and suddenly the error stops and everything is working fine. 
Then suddently the problem again occurs

## Versions

- ESPMQTTClient lib: [1.13.3](https://github.com/plapointe6/EspMQTTClient/releases/tag/1.13.3)
- PubSubClient lib: [v2.8](https://github.com/knolleary/pubsubclient/releases/tag/v2.8)
- Arduino ESP32 / ESP8622 core: -2.5.0- (To update it go to the arduino board manager and update ESP32 / ESP8266)

## Hardware
- Board type : ESP8266 ?
- Board model : Nodemcu 1.0

## C++ code

```c++
const String boardNumber = "071"; 
const String clientStr = "Client"+boardNumber;

#include "EspMQTTClient.h"
EspMQTTClient client(
  "IOT",
  "XXXXXXX",
  "192.168.195.249",  // MQTT Broker server ip
  //"MQTTUsername",   // Can be omitted if not needed
  //"MQTTPassword",   // Can be omitted if not needed
  clientStr.c_str(),     // Client name that uniquely identify your device
  1883              // The MQTT port, default to 1883. this line can be omitted  
);
```

EdJoPaTo, 2023-02-24:
Is this also an issue with the PubSubClient library directly? This library uses it for MQTT handling so I would assume this might be an issue over there.

---

Looking at your code I have some completely unrelated comment: I do like to use defines for strings as string concats are at compile time and a bit easier to write and handle compared to the `String` stuff that happens on execution:

```c
#define BOARD_NUMBER "071"
#define CLIENT_NAME "Client" BOARD_NUMBER

EspMQTTClient client(
 …
 CLIENT_NAME,
 …
);
```

benbecke, 2023-02-24:
i had a impementation before just using the pubsubclient lib. There i didn´t get this error :-/
But i had problems in handling the wifi therefore i switched to this one. 
But maybe i am wrong and i only didn´t see the error at all before... 

Ok thanks for this hint i will change that. 
Is there a way to direktly trace what messages are sent at mqtt connection time to see where the malformed value should be ?

EdJoPaTo, 2023-02-24:
you can enable `client.enableDebuggingMessages()` in the setup which shows the received and published MQTT messages via Serial

benbecke, 2023-02-27:
ok i did that before but its not showing the broker connection stuff. Only the publish and subscribe messages. 
Somehow i have to find out what is wrong in the initial connection message to the broker

Sperryfreak01, 2023-04-02:
I am running into similar issues, I am still narrowing down the issue but I see the same issue even if I don't construct the string.  client name ex: "sperryfreak01" -- I still get UTF 8 errors.  I think it has to do with some kind of memory being freed while the library is still pointed at it.  It only started for me when I switched over to the steps in #26

Sperryfreak01, 2023-04-03:
Well I figured it out, I was using "strings" (my python riddled brain isn't sure what is what anymore) as const char* and it was causing problems.  After converting everything to char var[] or char var[x] it all started working flawlessly.