plapointe6/EspMQTTClient #140: esp32 v3.x breaking changes - library doesn't compile for ESP32 boards

chrisweather, 2024-06-17:
## Description of the problem
With Espressif esp32 2.0.17 everything works fine.
esp32 3.x includes breaking changes to wifi handling 
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html

When I apply the changes in PR #138 manually this fixes the compiler error.
Didn't check if the library is full functionable.

Added 
#include <WiFi.h>
to line 22 in EspMQTTClient.h

## Hardware
- Board type : ESP32
- Board model : Lolin D32

## Logs
Arduino IDE 1.8.19 and 2.x
Espressif esp32 3.01

C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'bool EspMQTTClient::handleWiFi()':
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp:187:5: error: 'WiFi' was not declared in this scope
  187 |     WiFi.disconnect(true);
      |     ^~~~
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp:194:27: error: 'WiFi' was not declared in this scope
  194 |   bool isWifiConnected = (WiFi.status() == WL_CONNECTED);
      |                           ^~~~
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp:194:44: error: 'WL_CONNECTED' was not declared in this scope; did you mean 'MQTT_CONNECTED'?
  194 |   bool isWifiConnected = (WiFi.status() == WL_CONNECTED);
      |                                            ^~~~~~~~~~~~
      |                                            MQTT_CONNECTED
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp:215:27: error: 'WL_CONNECT_FAILED' was not declared in this scope; did you mean 'MQTT_CONNECT_FAILED'?
  215 |       if(WiFi.status() == WL_CONNECT_FAILED || millis() - _lastWifiConnectiomAttemptMillis >= _wifiReconnectionAttemptDelay)
      |                           ^~~~~~~~~~~~~~~~~
      |                           MQTT_CONNECT_FAILED
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'bool EspMQTTClient::handleMQTT()':
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp:325:9: error: 'WiFi' was not declared in this scope
  325 |         WiFi.disconnect(true);
      |         ^~~~
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'void EspMQTTClient::onWiFiConnectionEstablished()':
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp:362:75: error: 'WiFi' was not declared in this scope
  362 |       Serial.printf("WiFi: Connected (%fs), ip : %s \n", millis()/1000.0, WiFi.localIP().toString().c_str());
      |                                                                           ^~~~
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'void EspMQTTClient::onWiFiConnectionLost()':
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp:388:5: error: 'WiFi' was not declared in this scope
  388 |     WiFi.disconnect(true);
      |     ^~~~
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp: In member function 'void EspMQTTClient::connectToWifi()':
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp:555:3: error: 'WiFi' was not declared in this scope
  555 |   WiFi.mode(WIFI_STA);
      |   ^~~~
C:\ProgrammePortable\arduino-1.8.19\portable\sketchbook\libraries\EspMQTTClient\src\EspMQTTClient.cpp:555:13: error: 'WIFI_STA' was not declared in this scope; did you mean 'WIFI_IF_STA'?
  555 |   WiFi.mode(WIFI_STA);
      |             ^~~~~~~~
      |             WIFI_IF_STA

AdaBill, 2024-07-04:
Thanks chrisweather, I have been having trouble connecting with my ESP32 with the other libraries, just no response, then I saw this library and tried it and got the same WiFi problem.  I should learn to read the Bugs before I spend the time trying to fix something you already fixed.  Thanks Again. I also think that there should be an #include "PubSubClient.h" in the header to make sure that tis library is also installed.

cyijun, 2024-07-08:
Hi, you may try my library with similar interfaces. It works with the latest Arduino core.
https://github.com/cyijun/ESP32MQTTClient

steros76, 2024-07-26:
The solution is on
https://forum.arduino.cc/t/broken-dependencies/1266624/8

In "EspMQTTClient.h" you must add

  #include <WiFi.h>

before

  #include <WiFiClient.h>

chrisweather, 2024-07-26:
I know, that it can be fixed with this small change. There's also a PR for this fix for this library.
It would be more comfortable to simply install the library without any additional modifications. 
Unfortunately this project seems to be orphaned.

jonsummers, 2025-01-18:
This project orphaned? 6 years without an update and now broken.

How to fix for your platformio.ini

-  lib_deps = plapointe6/EspMQTTClient
+ lib_deps = https://github.com/zimbora/EspMQTTClient

teckdev, 2025-01-25:
Although this project is dead, it still works pretty well.
I forked it a while ago an just add the fix I need to it. This way I can be sure that my fork will always be there. 
Just  need to change your `lib_deps` to your fork as @jonsummers mentionned. i.e. `libdeps = https://github.com/<YOU>/EspMQTTClient`